ni/nifpga-python #6: Attribute Error on initialization of session py27 on win32

ellinorcrux, 2017-04-21:
I tried creating a new session and ran into an attribute error.
traceback:

```
 self._dummy_session = Session(bitfile, resource, reset_if_last_session_on_exit=False, no_run=True)
  File "C:\python27\Lib\site-packages\nifpga\session.py", line 85, in __init__
    self._nifpga = _NiFpga()
  File "C:\python27\Lib\site-packages\nifpga\nifpga.py", line 336, in __init__
    library_function_infos=library_function_infos)
  File "C:\python27\Lib\site-packages\nifpga\statuscheckedlibrary.py", line 179, in __init__
    func = getattr(library, lfi.name_in_library)  # i.e., dlsym()
  File "C:\python27\Lib\ctypes\__init__.py", line 378, in __getattr__
    func = self.__getitem__(name)
  File "C:\python27\Lib\ctypes\__init__.py", line 383, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: function 'NiFpgaDll_GetResourceIndex' not found

```
the Session object in this case is the one yielded by this import: 

from nifpga import Session

My python version is: 
Python 2.7.11 (v2.7.11:6d1b6a68f775, Dec  5 2015, 20:32:19) [MSC v.1500 32 bit (Intel)] on win32

draughoncollin, 2017-04-21:
Looks like it could be a complaint about the NiFpgaLv.dll missing. The NiFpgaLv.dll is part of the NI-RIO driver. Which version of NI-RIO is installed to your machine?

ellinorcrux, 2017-04-21:
The driver version appears to be NI RIO 15.0 

If it couldn't find the DLL, wouldn't it fail at these lines: 

```
 try:
      super(_NiFpga, self).__init__(library_name="NiFpga",
                                          library_function_infos=library_function_infos)
except LibraryNotFoundError as e:

```
This is in line 334 of nifpga.py

badgraphicsghost, 2017-04-21:
We fixed this error in a previous changelist, but I forgot to upload the package to pypi (I need to automate that process).  Try running:

    pip install nifpga==17.0.0.post2 -U

I just uploaded the latest master to pypi.

Your encountering this error because you are using a version of NI-RIO that is <16.0.  You can check the version of NI-RIO on your system under the Software tab in MAX.  I would recommend updating NI-RIO, previous versions of it should work with the Python API, but haven't been tested.  16.0 is the earliest version we officially support.

ellinorcrux, 2017-04-25:
I updated nifpga and my NI-RIO drivers. I actually have two different RIO drivers installed: NI-RIO 16.0 and Compact RIO 15.0 

Should this make any difference? 

I can now initialize, but the recommended initialization procedure from the docs doesn't work for me for several reasons. The docs say to do: 

```
with Session(bitfile="BitfilePath.lvbitx", resource="RIO0") as session:
   try:
      session.run()
   except FpgaAlreadyRunningWarning:
      pass
   session.download()
   session.abort()
   session.reset()
```
Firstly, I cannot import the exception FpgaAlreadyRunningWarning because I believe that is dynamically generated when it gets raised. So the best I can do is: 

`from nifpga.status import WarningStatus`

And then catch all WarningStatus child exceptions. I would really prefer not to do that. Could you please make the specific exception classes importable? 

Secondly, as soon as I get to the download step, I run into this exception: 

```
 File "C:\Python27\lib\site-packages\nifpga\session.py", line 171, in download
    self._nifpga.Download(self._session)
  File "C:\Python27\lib\site-packages\nifpga\status.py", line 85, in internal
    _raise_or_warn_if_nonzero_status(status, function_name, argument_names, args)
  File "C:\Python27\lib\site-packages\nifpga\status.py", line 51, in _raise_or_warn_if_nonzero_status
    raise codes_to_exception_classes[status](function_name, argument_names, *args)
nifpga.status.FpgaBusyFpgaInterfaceCApiError: Error: FpgaBusyFpgaInterfaceCApi (-61200) when calling 'NiFpgaDll_Download' with arguments:
	session: 0x10002L
```
Any idea why that might happen?

badgraphicsghost, 2017-04-25:
To your first point, we need to update those examples.  FpgaAlreadyRunningWarning was originally an exception.  Since then we have changed our Warnings to use python warnings.  We will get those updated shortly.

The following way of importing warnings should work.  Let me know if it doesn't.  The warnings are generated on module load, so they should be present.

    from nifpga.status import FpgaAlreadyRunningWarning

Related to FpgaBusyFpgaInterfaceCApiError, this error means we attempted to re-download the FPGA bitfile, but there were multiple Sessions open, so we couldn't.  We only allow download to be called when a single Session is open.  In the near future I intend to add in the error description strings to the exceptions, so it will be much clearer what the error means.

For initializing your device, the recommend flow generally is:

    with Session(bitfile="BitfilePath.lvbitx", resource="RIO0") as session:
        session.reset()  # resets the FPGA to the default state and stops it
        # TODO: write initial values to registers
        session.run() # starts the logic on the FPGA
        # logic that interacts with the FPGA.

Different flows can work for different applications though.

ellinorcrux, 2017-04-27:
If it's a warning then I won't be able to catch it. That's OK with me. I just want to be able to catch exceptions. I was able to import the FpgaBusyFpgaInterfaceCApiError exception. 

Does the FpgaBusyFpgaInterfaceCApiError mean that I can't have two sessions for two FPGA targets open at a time? 

Also I noticed that writing to an indicator does not raise an exception. Should I open another github issue for that? 

Thanks!

badgraphicsghost, 2017-04-27:
Warnings are kinda catchable, you can do something like we do [here](https://github.com/ni/nifpga-python/blob/master/nifpga/tests/test_nifpga.py#L110).  Overall though, we have designated them as warnings because its safe to ignore them.

FpgaBusyFpgaInterfaceCApiError just means that on a single device, there was more than one Session open and you attempted to call the "download" function.  Calling the "download" function requires exclusive access to that device.  

There are no limits with regards to using multiple FPGA Targets.  

Multi-session, multi-threaded, and multi-process access to a single FPGA Target is also fine.  There are two caveats on that though:
1. You can only call the "download" function when a single Session is open for that FPGA Target.  This includes Sessions from other sources such as LabVIEW or the C API.
2. The first time you use a FIFO, it will be joined to the Session it is used from.  Other Sessions cannot use that FIFO until that first joined Session is closed.

Thanks for the questions.  I might use these to start a FAQ section.

ellinorcrux, 2017-04-27:
Great!

closed 2017-04-27 by ellinorcrux