ni/nifpga-python #10: Request for nifpga to support more LabVIEW datatypes

cgiustini, 2017-09-20:
Hi,

The current API allows Python code to read/write data to/from the running FPGA bitfile only when the data in question is made of very simple types (integers, booleans). The Python API doesn’t let the user interact with the bitfile for any of the more “complex” data types the FPGA bitfile may expose (for instance: fixed point number, a cluster representing the state of something on the FPGA). For most advanced use-cases where the bitfiles expose such data types, the nifpga Python module as-is cannot integrate with the existing NI FPGA code. Is there any plan to have nifpga support more types? I'd see a lot of value in adding such a feature.

Thanks,
Carlo

badgraphicsghost, 2017-09-20:
Hi Carlo

Support for more LabVIEW datatypes is our next planned feature.  Currently it looks like we will get to it either later this year or early next year.

In the meantime, the options for accessing the data are:
1. Convert it or pack it into a supported type on the FPGA.
2. Maybe check out [mmatthe's fork](https://github.com/mmatthe/nifpga-python).  I know he was looking into adding support for FXP and Cluster types, but I'm not sure what the state of his branch is.  I'll be taking a closer look at his branch once we start working on the feature.

Thanks for the feedback,
Michael

cgiustini, 2017-09-25:
Thank you. Ok sounds good. I think that that officially implementing/supporting this feature would be really excellent.

closed 2017-09-25 by cgiustini