ni/nifpga-python #11: System error when using pyinstaller

badrobit, 2017-12-18:
**OS:** Windows10 64bit 
**Labview:** 2017
**Labview Drivers:** May2017
**nifpga:** 17.0.0.post2

The program runs fine when run from the command line. When I attempt to bundle it together into an executable for use I get the following error on the command line: 

```bash
Fatal Python error: Py_Initialize: unable to load the file system codec
LookupError: unknown encoding: utf-8

Current thread 0x000037c8 (most recent call first):
```

I have tried removing all other imports and without nifpga everything works fine. Once I add the nifpga package into the mix I get this error.

TJ-G, 2017-12-18:
Hey Badrobit,

I'm the project manager for the NI Group that owns this api - just wanted to let you know that we likely won't be able to take a look into this properly until after the holidays. My apologies if this is a big issue in the short-term, but we'll take a look as soon as we can.

badrobit, 2017-12-18:
No problem, I appreciate you taking a look at it, please let me know if there is anything else that you need from me to help sort out this behaviour.

mgumble, 2018-01-03:
Badrobit,

I have failed to reproduce this error using a simple python file importing nifpga, and matching the system you described with Python 2.7. I was curious what version of python you were using, and if you can reproduce this problem for a simple program (including nifpga) or just the application you are working on.

badrobit, 2018-01-17:
I was using python 3.6 could this be causing the issue?

badgraphicsghost, 2018-06-26:
This issue should be resolved in 18.1.0.  Please re-open if thats not the case.

closed 2018-06-26 by badgraphicsghost