ni/nifpga-python #35: Occasionally Error: RpcServerMissing (-63044) when calling 'NiFpgaDll_Open'

BrankoIlic, 2020-05-28:
I'm running unit tests, and after about fourty reads from cRio (by fourty Sessions) I start to get this error:

 File "C:\Program Files (x86)\Jenkins\workspace\E-TEST\Test\helpers\motor_monitor_helper.py", line 30, in get_current_position
    with Session(self.file, self.resource) as session:
  File "C:\Python27\lib\site-packages\nifpga\session.py", line 109, in __init__
    self._session)
  File "C:\Python27\lib\site-packages\nifpga\status.py", line 100, in internal
    _raise_or_warn_if_nonzero_status(status, function_name, argument_names, args)
  File "C:\Python27\lib\site-packages\nifpga\status.py", line 66, in _raise_or_warn_if_nonzero_status
    raise codes_to_exception_classes[status](function_name, argument_names, *args)
RpcServerMissingError: Error: RpcServerMissing (-63044) when calling 'NiFpgaDll_Open' with arguments:
        bitfile path: 'b'C:\Program Files (x86)\Jenkins\workspace\E-TEST\Test\NiFpga_eTest.lvbitx''
        signature: 'b'1EEE087A3560B2C0A53EF8014DCE226E''
        resource: 'b'rio://10.0.4.89/RIO0''
        attribute: 0x0
        session: 0x0L

That line 30 of motor_monitor_helper.py is:
29    def get_current_position(self):
30        with Session(self.file, self.resource) as session:
31            Settings.get_instance().load()
32            positionCounts = session.registers['Position (Counts)'].read()
33            print("session.registers['Position (Counts)']: {}".format(positionCounts))
34            return float(positionCounts) / float(Settings.get_instance().encoder_resolution)

And after that I have to reset cRio to make it works again.
So, all that 40 reads go fine, and that it stops. I have changed tests that are running, but it depends only of number of tests, not of tests itself.

My python version is 2.7.16
nifpga is Version: 19.0.0

badgraphicsghost, 2020-05-28:
What cRIO model are you using?  Also what version of NI-RIO is on the cRIO and on the Windows machine (you can view this in MAX under software)?

Do you have 40 sessions open at once or are you opening and closing a bunch of sessions?

BrankoIlic, 2020-05-29:
Big thanks for quick response :)
cRIO is 9025 
NI-RIO on the cRIO is 19.1, but there is also line "NI CompactRIO 19.5 - September 2019"
NI-RIO on the Windows machine is 19.5.0

Sessions are open and closed sequentially with time distance, never more at once.

badgraphicsghost, 2020-05-29:
You mention that you need to reset the cRIO to make it work again.  Could you try leaving it idle for a few minutes after it gets into that state and seeing if it recovers?  You might also check serial out when the problem occurs.

This sounds a lot like socket exhaustion which our older vxworks targets were prone to running into at times.  If that ends up being the case, then there might be a workaround.

BrankoIlic, 2020-06-02:
As I said, it works fine now. 
And to answer your question: I can not change it now, but as I recall it has been sometimes idle for few minutes but it would not recover.

closed 2021-06-10 by badgraphicsghost