ni/nifpga-python #38: Registers ending with a whitespace are not shown correctly on the registers.keys()

VicidominiLab, 2022-07-07:
When a FPGA LabView code has been compiled with a register with a name ending with a whitespace like( `"MyRegister " `) is not correctly reported in session.registers.keys() .
`print(session.registers.keys())`
shows a
`dict_keys([''MyRegister"])` 
instead of 
`dict_keys([''MyRegister "])`

and even if it is reported wrongly the
`session.registers[''MyRegister"].write(0.0)`
throw an error 
`KeyError(''MyRegister")`

strainmike, 2022-07-28:
What version of python are you using?

I wasn't able to reproduce this, but when searching, I did find a few python bugs in different versions related to the xml parser handling whitespace incorrectly.

strainmike, 2022-08-10:
Someone pointed out to me that the XML spec ignores leading and trailing whitespace in tags.  Unfortunately LabVIEW just stores register names (including whitespace) without caring about that.  Thats likely the issue.

Unfortunately I don't see any easy solutions or flags we could pass into the parser to always include the whitespace.  I'm going to leave this issue open for now.