ni/nifpga-python #43: Error: InvalidResourceName (-63192) when calling 'NiFpgaDll_Open

manishkumaretn, 2023-07-27:
Hi. I am getting following error while opening the session with NIFPGA:

nifpga.status.InvalidResourceNameError: Error: InvalidResourceName (-63192) when calling 'NiFpgaDll_Open' with arguments:
	bitfile path: b'C:\\Eclipse_WS\\NIDAQMX\\SquareNumber_FPGATarget_Test1_okVPBnie6P8.lvbitx'
	signature: b'D9DA997A0C0155146DD77DAEC8CC6F9
[image]
[image]
7'
	resource: b'RIO1'
	attribute: 0x0
	session: 0x0

I cross checked the Resource name in NI-MAX and same name is updated in Target file as well. Pl find attached image of MAX and Project properties.

Test environment details:
3.6.0 (v3.6.0:41df79263a11, Dec 23 2016, 07:18:10) [MSC v.1900 32 bit (Intel)]
Package    Version
---------- -------
future     0.18.3
nidaqmx    0.5.7
nifpga     22.0.0
numpy      1.19.5
pip        21.3.1
setuptools 28.8.0
six        1.16.0

Thanks,
Manish

strainmike, 2023-07-27:
Since you are talking to a device on a remote PXI target, you need to change your resource name from "RIO1"to "rio://192.168.1.1/RIO1".  That will tell that tell NiFpga to open a remote session to RIO1 on 192.168.1.1 instead of opening a session to a device named RIO1 on your windows machine.

manishkumaretn, 2023-07-31:
Thanks for your response. It is working now.

closed 2023-07-31 by strainmike