ni/nifpga-python #46: OSError: exception: access violation reading 0x0000000000000008

lunarspectrum, 2024-06-10:
# Description
`OSError: exception: access violation reading 0x0000000000000008` occurs when trying to load a simple bitfile. 

# System Info
python 3.12.3 (64-bit, via miniconda)
nifpga 22.0.0

Windows 10 (64 bit)
NI USB-7845R
NI R Series Multifunction RIO Software version 18.5
VS

# Steps to reproduce
Using a simple example bitfile that runs using LabView, run 
```Python
from nifpga import Session
with Session(bitfile=r"my_simple_bitfile.lvbitx", resource="RIO0") as session:
    session.reset()
    session.run()
```

# Output 
```
Traceback (most recent call last):
  File "test_nifpga.py", line 3, in <module>
    with Session(
         ^^^^^^^^
  File "<environment path>\Lib\site-packages\nifpga\session.py", line 118, in __init__
    self._nifpga.Open(bitfile_path,
  File "<environment path>\Lib\site-packages\nifpga\status.py", line 99, in internal
    status = function(*args)
             ^^^^^^^^^^^^^^^
OSError: exception: access violation reading 0x0000000000000008
```

# Steps taken to resolve
Reinstalled and updated python from earlier version (was 3.10.x). Repaired NI installation. Contacted IT to verify that security software is not blocking access to the DLLs. Tried a different bitfile that runs in LabView. Reviewed [this](https://knowledge.ni.com/KnowledgeArticleDetails?id=kA0VU0000000eqH0AQ&l=en-US) NI KB article.

strainmike, 2024-06-10:
Just to confirm, it sounds like Python 3.10.x works for you, but upgrading to Python 3.12.x was broken?  Or am I reading that wrong?

lunarspectrum, 2024-06-10:
No. It does not work under either version I have tried. I initially encountered the error in 3.10.

strainmike, 2024-06-10:
Our ATS currently runs with Python 3.9, but I manually ran it with 3.12 and couldn't reproduce any issues.  

I hadn't ever seen that KB before but clearly someone has run into this problem before.  I can't find any internal docs about it though.

Could you try running something similar to the following to try to narrow down the issue:

    import ctypes
    
    bitfile = "my_simple_bitfile.lvbitx"
    resource = "RIO0"
    
    nifpga = ctypes.CDLL('NiFpga.dll')
    
    nifpga.NiFpgaDll_Open.argtypes = [
        ctypes.POINTER(ctypes.c_char),  # bitfile
        ctypes.POINTER(ctypes.c_char),  # signature
        ctypes.POINTER(ctypes.c_char),  # resource
        ctypes.c_uint32,                # attribute
        ctypes.POINTER(ctypes.c_uint32) # session
    ]
    
    nifpga.NiFpgaDll_Open.restype = ctypes.c_int32
    
    def open_session(bitfile, resource):
        session = ctypes.c_uint32()
        result = nifpga.NiFpgaDll_Open(
            ctypes.c_char_p(bitfile.encode('utf-8')),
            None,
            ctypes.c_char_p(resource.encode('utf-8')),
            ctypes.c_uint32(1 << 31),
            session
        )
        return result, session.value
        
    print(open_session(bitfile, resource))

This should succeed and isn't doing anything too special or nifpga specific.  If that fails, you might try some very basic CDLL calls.

Note, I'm about to be OOO for a few weeks.  If you need support before then you will probably need to open a ticket on https://www.ni.com/en/support.html

lunarspectrum, 2024-06-12:
Hi! Thanks for the quick response. Alas, this gives the same error message. 

Also code that does not require a bitfile:
```python
import ctypes
import nifpga
lib = nifpga.nifpga._NiFpga()
resourceIndex = ctypes.c_uint32()
resourceName = ctypes.c_char_p(b"RIO0")
lib.GetResourceIndex(resourceName,ctypes.byref(resourceIndex))
```
also gives
```
---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
Cell In[8], line 1
----> 1 lib.GetResourceIndex(resourceName,ctypes.byref(resourceIndex))

File <environment path>\Lib\site-packages\nifpga\status.py:99, in check_status.<locals>.decorator.<locals>.internal(*args)
     96 if hasattr(function, "argtypes") and len(args) != len(function.argtypes):
     97     raise TypeError("%s takes exactly %u arguments (%u given)"
     98                     % (function_name, len(function.argtypes), len(args)))
---> 99 status = function(*args)
    100 _raise_or_warn_if_nonzero_status(status, function_name, argument_names, args)

OSError: exception: access violation reading 0x0000000000000008

```

I've also updated the fpga dll to 19.0.0.49152, which has the same results. 

I will contact support. Thank you again.

lunarspectrum, 2024-06-13:
Ok, I've got this and the original code working under python 3.10.4, 32-bit. I would *really* like to get this to work under a non-deprecated version. I've been in contact w/ support.

lunarspectrum, 2024-07-04:
Using a new computer with a fresh installation of everything. Still getting the same error message.

lunarspectrum, 2024-07-09:
I seem to have solved it by looking more closely at anti-virus software. What a bizarre bug.

closed 2024-07-09 by lunarspectrum

strainmike, 2024-07-09:
@lunarspectrum sorry meant to get back to this and ask more questions on how to reproduce it.  If possible, could you provide any more details for anyone else that runs into the issue.

Thanks!